facchinm/avrdude #3: Release plan

lollinopil, 2018-08-20:
Dear contributors, as you know the latest AVRdude release (6.3) does not support the ATtiny 1-series  family.

I am about to start dealing with an ATtiny412 MCU and I noticed that your current AVRdude code is able to handle it.  Is it planned to be released? If so, do you have any idea of the timing?

Thank you for any help!

facchinm, 2018-08-20:
Hi @lollinopil ,
the current avrdude (packaged in Arduino Hourly and Beta) already contains the definitions for ATTiny412 (https://github.com/facchinm/avrdude/blob/master/avrdude.conf.in#L15670); is there something missing from our side? Avrdude's SVN looks offline at the moment so I cannot check if any other commit has slipped in after my last clone.

facchinm, 2018-08-20:
Relevant patch is https://github.com/facchinm/avrdude/commit/4757842c0f65eac13c42ca2f8d81d0626e41a016

lollinopil, 2018-08-20:
Dear Facchinm, thank you very much for your reply! Nothing missing from your side, probably i just don't know how to upgrade my toolchain. 

I currently have winAVR installed on my PC and linked to my AtmelStudio. Shall I just replace avrdude.conf file into its *.bin folder? If not, is there any guide to follow? 

Thank you for your help.

lollinopil, 2018-08-23:
Yes, it is exactly as you said. Thank you very much!

danupp, 2019-06-19:
Following this thread, what is the plan for supporting Attiny817 and similar new chips?

facchinm, 2019-06-19:
`avrdude-6.3.0-arduino17` contains support for all new Attinies (PR here https://github.com/arduino/Arduino/pull/8976 ).
Any testing is important to validate the toolchain soon :wink:

danupp, 2019-06-19:
Thank you for your reply! So there is a parallel branch somewhere for avrdude? (Why not in this repo?) 
I tried build 864 in that thread but there doesn't seem to be such Attinies support(?). The binary in hardware/tools/avr/bin also still says version 6.3-. ?

facchinm, 2019-06-19:
Maybe I'm not getting your question; are you asking for support in `avrdude` or an Arduino target? The latter case in not going to happen and you have to use a third party core (if it exists).
In the first case, the avrdude contained there supports new Attinies via UPDI programming and is indeed derived from this master branch ( https://github.com/facchinm/avrdude/blob/master/avrdude.conf.in#L15818 ).
Thanks for spotting the problem with version name, I'm fixing it immediately

danupp, 2019-06-20:
I am not interested in Arduino but want an avrdude version that lets me load compiled software into the newest attinies over the UPDI interface. Using for instance the programmer on the ATtiny817xplainedmini. Sorry if I was not clear.
When running avrdude in the build 864 with '-p t817' it says part not found and shows a list with a lot of AVR's but not he latest ones.

danupp, 2019-06-25:
Maybe there is a better place to interact with the developers of avrdude? I don't completely understand why this repo exists when the official site just links to a svn repo. Thank you.

facchinm, 2019-06-25:
@danupp the place you are looking for is https://savannah.nongnu.org/projects/avrdude . Unfortunately Joerg has not published an official release since 2016 so we have to maintain this "soft fork" (mirror + couple of patches applied).
About the missing '-p t817' , which operating system are you using? Can you paste here the commandline you are trying to run? Because 6.3.0-arduino17 has the entry for t817 on every package I checked.

danupp, 2019-06-25:
Thank you for info. Why is the program bundled with Arduino? Wouldn't it be better to "release" avrdude in its own versions?
 I am using Linux Mint. Could it be that a config file is missing (or evidently some path variable hard coded)?

_d@linux-HP:~/Downloads/arduino-PR-8976-BUILD-864/hardware/tools/avr/bin$ ./avrdude -t t817 -c xplainedmini
avrdude: can't open config file "/home/jenkins/workspace/avrdude-staging/label/debian7-x86_64/objdir/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/home/jenkins/workspace/avrdude-staging/label/debian7-x86_64/objdir/etc/avrdude.conf"_

When I run with the conf file from the avrdude package in mint I instead get:

_d@linux-HP:~/Downloads/arduino-PR-8976-BUILD-864/hardware/tools/avr/bin$ ./avrdude -C /etc/avrdude.conf -t t817 -c xplainedmini
avrdude: No AVR part has been specified, use "-p Part"

Valid parts are:
  uc3a0512 = AT32UC3A0512
  c128     = AT90CAN128
  c32      = AT90CAN32
  c64      = AT90CAN64
  pwm2     = AT90PWM2
  pwm216   = AT90PWM216
  pwm2b    = AT90PWM2B
  pwm3     = AT90PWM3
  pwm316   = AT90PWM316
  pwm3b    = AT90PWM3B
  1200     = AT90S1200
  2313     = AT90S2313
  2333     = AT90S2333
  2343     = AT90S2343
  4414     = AT90S4414
  4433     = AT90S4433
  4434     = AT90S4434
  8515     = AT90S8515
  8535     = AT90S8535
  usb1286  = AT90USB1286
  usb1287  = AT90USB1287
  usb162   = AT90USB162
  usb646   = AT90USB646
  usb647   = AT90USB647
  usb82    = AT90USB82
  m103     = ATmega103
  m128     = ATmega128
  m1280    = ATmega1280
  m1281    = ATmega1281
  m1284    = ATmega1284
  m1284p   = ATmega1284P
  m1284rfr2 = ATmega1284RFR2
  m128rfa1 = ATmega128RFA1
  m128rfr2 = ATmega128RFR2
  m16      = ATmega16
  m161     = ATmega161
  m162     = ATmega162
  m163     = ATmega163
  m164p    = ATmega164P
  m168     = ATmega168
  m168p    = ATmega168P
  m168pb   = ATmega168PB
  m169     = ATmega169
  m16u2    = ATmega16U2
  m2560    = ATmega2560
  m2561    = ATmega2561
  m2564rfr2 = ATmega2564RFR2
  m256rfr2 = ATmega256RFR2
  m32      = ATmega32
  m324p    = ATmega324P
  m324pa   = ATmega324PA
  m325     = ATmega325
  m3250    = ATmega3250
  m328     = ATmega328
  m328p    = ATmega328P
  m329     = ATmega329
  m3290    = ATmega3290
  m3290p   = ATmega3290P
  m329p    = ATmega329P
  m32m1    = ATmega32M1
  m32u2    = ATmega32U2
  m32u4    = ATmega32U4
  m406     = ATMEGA406
  m48      = ATmega48
  m48p     = ATmega48P
  m48pb    = ATmega48PB
  m64      = ATmega64
  m640     = ATmega640
  m644     = ATmega644
  m644p    = ATmega644P
  m644rfr2 = ATmega644RFR2
  m645     = ATmega645
  m6450    = ATmega6450
  m649     = ATmega649
  m6490    = ATmega6490
  m64rfr2  = ATmega64RFR2
  m8       = ATmega8
  m8515    = ATmega8515
  m8535    = ATmega8535
  m88      = ATmega88
  m88p     = ATmega88P
  m88pb    = ATmega88PB
  m8u2     = ATmega8U2
  t10      = ATtiny10
  t11      = ATtiny11
  t12      = ATtiny12
  t13      = ATtiny13
  t15      = ATtiny15
  t1634    = ATtiny1634
  t20      = ATtiny20
  t2313    = ATtiny2313
  t24      = ATtiny24
  t25      = ATtiny25
  t26      = ATtiny26
  t261     = ATtiny261
  t28      = ATtiny28
  t4       = ATtiny4
  t40      = ATtiny40
  t4313    = ATtiny4313
  t43u     = ATtiny43u
  t44      = ATtiny44
  t441     = ATtiny441
  t45      = ATtiny45
  t461     = ATtiny461
  t5       = ATtiny5
  t84      = ATtiny84
  t841     = ATtiny841
  t85      = ATtiny85
  t861     = ATtiny861
  t88      = ATtiny88
  t9       = ATtiny9
  x128a1   = ATxmega128A1
  x128a1d  = ATxmega128A1revD
  x128a1u  = ATxmega128A1U
  x128a3   = ATxmega128A3
  x128a3u  = ATxmega128A3U
  x128a4   = ATxmega128A4
  x128a4u  = ATxmega128A4U
  x128b1   = ATxmega128B1
  x128b3   = ATxmega128B3
  x128c3   = ATxmega128C3
  x128d3   = ATxmega128D3
  x128d4   = ATxmega128D4
  x16a4    = ATxmega16A4
  x16a4u   = ATxmega16A4U
  x16c4    = ATxmega16C4
  x16d4    = ATxmega16D4
  x16e5    = ATxmega16E5
  x192a1   = ATxmega192A1
  x192a3   = ATxmega192A3
  x192a3u  = ATxmega192A3U
  x192c3   = ATxmega192C3
  x192d3   = ATxmega192D3
  x256a1   = ATxmega256A1
  x256a3   = ATxmega256A3
  x256a3b  = ATxmega256A3B
  x256a3bu = ATxmega256A3BU
  x256a3u  = ATxmega256A3U
  x256c3   = ATxmega256C3
  x256d3   = ATxmega256D3
  x32a4    = ATxmega32A4
  x32a4u   = ATxmega32A4U
  x32c4    = ATxmega32C4
  x32d4    = ATxmega32D4
  x32e5    = ATxmega32E5
  x384c3   = ATxmega384C3
  x384d3   = ATxmega384D3
  x64a1    = ATxmega64A1
  x64a1u   = ATxmega64A1U
  x64a3    = ATxmega64A3
  x64a3u   = ATxmega64A3U
  x64a4    = ATxmega64A4
  x64a4u   = ATxmega64A4U
  x64b1    = ATxmega64B1
  x64b3    = ATxmega64B3
  x64c3    = ATxmega64C3
  x64d3    = ATxmega64D3
  x64d4    = ATxmega64D4
  x8e5     = ATxmega8E5
  ucr2     = deprecated, use 'uc3a0512'_

(Of course there should also be a program or verify option in the command given.)

facchinm, 2019-06-25:
You should use the config bundled with that version of avrdude, not the system wide one
```
d@linux-HP:~/Downloads/arduino-PR-8976-BUILD-864/hardware/tools/avr/bin$ ./avrdude -p t817 -c xplainedmini -C ../etc/avrdude.conf
```
will do the trick. About the distribution, avrude itself is shipped as a standalone package, we simply don't provide the download links but you can easily grab them from https://downloads.arduino.cc/packages/package_index.json

closed 2019-06-25 by facchinm

danupp, 2019-06-26:
@facchinm , thank you very much for your help. 
However I don't understand how I can get avrdude from that link(?)

Also I get the following response now. Any ideas? Sudo for full access. Don't understand why it thinks the board is busy. Is the program enable instruction error critical?

```
d@linux-HP:~/Downloads/arduino-PR-8976-BUILD-864/hardware/tools/avr/bin$ sudo ./avrdude -C ../etc/avrdude.conf -p t817 -c xplainedmini
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy

avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
avrdude: stk500v2_program_enable(): program enable instruction not defined for part "ATtiny817"
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
avrdude: jtag3_edbg_signoff(): unexpected response 0x03, 0x80

avrdude done.  Thank you.
```

Thanks!

danupp, 2019-06-26:
Got some help from this thread: https://www.avrfreaks.net/forum/xplained-nano-medbg-no-connection
When rebuilding avrdude with libhidapi and changing from xplainedmini to xplainedmini_updi I am now down to:

`avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0
`

I did not find a libhid to install, just libhidapi. Could this be part of the problem?

matt-allan, 2021-02-13:
@danupp (or anyone reading this with the same issue) I had the same issue and was able to resolve it by applying [this patch](https://lists.nongnu.org/archive/html/avrdude-dev/2019-01/msg00006.html) to the HEAD of this fork.

danupp, 2021-02-13:
> @danupp (or anyone reading this with the same issue) I had the same issue and was able to resolve it by applying [this patch](https://lists.nongnu.org/archive/html/avrdude-dev/2019-01/msg00006.html) to the HEAD of this fork.

Thanks matt-allen! I did however migrate to using pyUPDI instead, for programming newer AVR devices with the UPDI interface. That is much easier than struggling with building avrdude. But a pity.