facchinm/avrdude #12: Support for extended UPDI + new UPDI programmers

MCUdude, 2020-12-15:
This patch is a _big deal_ for us who develop for mega0/tiny0/tiny1/tiny2 and AVR-DA/DB/DD/EA. We finally have Avrdude support for the Pickit4 (UPDI), SNAP programmer! Would you be so nice and add this to the Arduino version so "modern" UPDI programmers can be used with Arduino/Avrdude? Thanks!

facchinm, 2020-12-16:
I think you refer to #11 , correct? I'm going to start a build this afternoon and probably publish a megaAVR core bundling it by next week.

MCUdude, 2020-12-16:
I wasn't even aware that #11 existed. He was just a few hours quicker than me 😉 

But yes, a release with #11 included would be great!

facchinm, 2020-12-16:
Closing since #11 got merged and packages rebuilt

closed 2020-12-16 by facchinm