facchinm/avrdude #13: JTAGICE3/Atmel ICE fails to verify lock byte due to reverted commit

matthijskooijman, 2020-12-29:
@facchinm, in 6b0863177b65f70d14e68036a10da92ddb122473 you reverted an upstream commit that fixed an issue with verifying lock bytes on the JTAGICE3 and Atmel ICE programmers. See https://github.com/arduino/Arduino/issues/11107

The commit message does not show any rationale for reverting this, do you still know why it was reverted? Can we unrevert it?

To reproduce, run the following (for a JTAGICE3 with an Uno attached on ISP, for atmelice the `-p` option must be changed):

```
$ ~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -cjtag3isp -Ulock:w:0x0F:m
[ Snip boring output]
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as DE
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)

avrdude done.  Thank you.

```

matthijskooijman, 2020-12-29:
I just tried compiling master to confirm it shows the problem, and reverted that revert to confirm it fixes the problem on JTAGICE3.

facchinm, 2020-12-30:
Thanks for investigating @matthijskooijman ! As I wote in the other thread, the history on this fork is a bit of a mess, so I'd rather spend a day making sure the SVN trunk is fine, applying the relevant patches and publishing again without all the reverts and stuff (of course on another branch :slightly_smiling_face:  )