facchinm/avrdude #18: Avrdude always needs the absolute path to avrdude.conf

kristofmulier, 2021-12-14:
Whenever one launches `avrdude`, it seems like the full absolute path to the `avrdude.conf` file must be given with the `-C` argument. I tried with a relative path (relative to the `avrdude` executable, something like `../etc/avrdude.conf`) but that doesn't work. Avrdude absolutely wants the *absolute* path.

In the makefile that we created for Embeetle IDE, we therefore end up with monstruous expressions, like:

```makefile
FLASHTOOL_PATH = $(if $(filter $(dir .),$(dir $(FLASHTOOL))),$(firstword $(wildcard $(patsubst %,%/$(FLASHTOOL),$(subst $(PATHSEP), ,$(PATH))))),$(FLASHTOOL))

.PHONY: flash
flash: $(BINARIES) print_flash
	"$(FLASHTOOL)" -C "$(subst /bin/../,/,$(subst /avrdude\$,,$(FLASHTOOL_PATH)\$)/../)etc/avrdude.conf" \
                 -v \
                 -patmega328p \
                 -carduino \
                 -P$(FLASH_PORT) \
                 -b115200 \
                 -D \
                 -Uflash:w:$(ELF_FILE:.elf=.hex):i
```

It would be so much easier if one could pass a *relative* path in the `-C` argument. Or even if the `-C` argument could be left out alltogether and `avrdude` looks itself for the file at `../etc/avrdude.conf`.

@facchinm - *the issue raised above is not a criticism. I am very thankful for your time and efforts to maintain avrdude. It's a great piece of software, and people like you are my heroes.*

kristofmulier, 2021-12-20:
Hi @facchinm,

Of course, I don't want to take advantage of your goodness (under the pretext of opensource). People like you deserve a statue. Least I can do is give a fair reward for your efforts. I would like to make a donation first, before you even look into the avrdude issue I raised above. Can I forward 150 euro to your PayPal account? Or through some other means? I'm flexible ^_^

Also, my apologies I didn't offer this earlier ^_^

Kind regards,
Kristof

MCUdude, 2021-12-21:
Now that the Avrdude project has migrated to Github, you can open an issue there
https://github.com/avrdudes/avrdude/

If you want to test on a pre-compiled, "official" binary, download the 6.4 version, which was released a few days ago:
https://download.savannah.gnu.org/releases/avrdude/

Future releases will be hosted on Github I'm sure.

kristofmulier, 2021-12-21:
Thank you @MCUdude for referring me to that GitHub page. I posted the issue there.