facchinm/avrdude #19: Arduino's "Avrdude fork" and Avrdude v7.0

MCUdude, 2022-04-06:
Hi @facchinm and @matthijskooijman!

Ever since the Avrdude project was moved over from Savannah to Github, it has seen numerous bug fixes and improvements.
One of the most important ones is support for "pyupdi/pymcuprog" style UPDI programming, where a USB to serial adapter can work as a full-blown UPDI programmer with just an external resistor. A new major release, Avrdude 7.0 is soon to be released, and it would be great if Arduino could start bundling 7.0, or at least a fork, where all [7.0 features](https://github.com/avrdudes/avrdude/blob/7e26a153752c9ebf346bd4e5db2e7c9b78a8b3c6/NEWS#L8-L323) are present.

I believe this repo exists because the Avrdude project was "dead" for several years, and you needed a way to fix reported issues. However, the commit history in this repo is a bit of a mess and scattered over several branches, where some of the commits from Savannah were later reverted without explaining why.

I was hoping that you perhaps could put in a little effort and copy over some of the "Arduino improvements" over to the [official Avrdude repo](https://github.com/avrdudes/avrdude). I'll be glad to help, but I'm having a real hard time keeping track of which "Savannah commits" have been included, ignored, or reverted, or which commits are real improvements that never made it back to Savannah.

Thanks!

MCUdude, 2022-04-06:
To sum up my somewhat vague question and add a few others:

* Which branch in this repo are you building Avrdude from? Is it `mainline`? If yes, why does it [differ so much from the other branches](https://github.com/facchinm/avrdude/branches/all)?
* Do you have a list of the various changes introduced in each `avrdude 6.3-arduinoN` version so I can help out getting these fixes/improvements back to the main repo?
* When looking at the history for the various branches, commits like b6fc553, 6b109f4, e202cc7, 6b08631 and 688813c are just undoing patches without explaining why. I'd be curious to know why, so we instead might fix the underlying issue.

matthijskooijman, 2022-04-06:
@MCUdude Thanks for your interest, I would be very glad if Arduino could be shipping an unmodified avrdude version indeed.

However, I'm not quite involved in this repo, so I'll have to defer to @facchinm for your questions.

facchinm, 2022-04-07:
Hi @MCUdude ,
closing this repo and posting all the relevant patches to the official avrdude's revamped home is still one of my wildest dreams :slightly_smiling_face: .
Unfortunately I'm not involved anymore in tool packaging inside Arduino; the request to provide upstream builds and create the PRs is in @arduino/team_tooling backlog but I can't give an estimation of its deadline.

About the patches, some of them were only needed by the Yun peculiar setup, while others were part of a strange project to be able to program both 16u2 and 328p with the same programmer on a Uno.

The reason why the branches diverged so much is that we were proving prebuilts for megaAVR (atmega4809 and friends) while the feature was not yet mainlined; when it finally happened, the branches had diverged so much that backporting the new relevant patches (with the extra svn-to-git passage) was easier than rebasing on master.

MCUdude, 2022-04-09:
Thanks for your reply @facchinm!

When looking at the Arduino avrdude changelog and comparing it with the official Avrdude repo, pretty much all improvements have made it back to the official repo. Do you think it would be possible to convince the tooling team to ship Avrdude 7.0 once released, with modifications if needed? For developers like me that are providing 3rd party Arduino cores, the 7.0 release is a _big deal_.

My favorites:
* Support for _way_ higher baud rates than 115200 when using jtag2updi
* Support for the Teensy bootloader
* Support for the Micronucleus (Digispark) bootloader
* Support for SerialUPDI (Pymcuprog/Pyupdi)
* DTR auto-reset performed on every retry when not able to communicate with the `-c arduino` programming option
* "Custom" baud rates like 128000, 250000, 500000, etc. finally supported on macOS

per1234, 2022-04-10:
Thanks so much for your initiative @MCUdude, and for the excellent work you have been doing on AVRDUDE!

I can comment on the little piece of this mess that I had some involvement with:

## Background

Way back in 2013, @kcuzner implemented a `linuxspi` programmer for AVRDUDE ([documented](https://web.archive.org/web/20210419190936/http://kevincuzner.com/2013/05/27/raspberry-pi-as-an-avr-programmer/), [patch](https://savannah.nongnu.org/patch/?8228)). This awesome thing allows you to use the native SPI interface of a Linux SBC as an efficient programmer.

Upstream AVRDUDE was not quick to act on this, but Arduino pulled it into their build in 2016 (https://github.com/facchinm/avrdude/commit/d7eacf839c4e040b155b2903f605d94201fc5463).

It came to my attention via a request to add it as one of the official programmer options: https://github.com/arduino/Arduino/issues/7795

While I was experimenting with that possibility, I discovered there was a problem with the implementation that made it unusable in that application (https://github.com/kcuzner/avrdude/issues/10). I submitted a pull request (https://github.com/kcuzner/avrdude/pull/17) to the repository where the patch was staged with the thought that the patch author would be the most appropriate to decide on whether it was the right approach, and if so to push that into the upstream submission. That was accepted and so I also submitted it to Arduino's AVRDUDE code base, since it already contained the original linuxspi patch that contained the bug.

A patch for a revised implementation of the linuxspi programmer was later submitted upstream (https://savannah.nongnu.org/patch/?9816) and this was eventually merged in https://github.com/avrdudes/avrdude/commit/40b0b104d6967ca75929706996b3dede304b9b25.

## Conclusion

I have tested the upstream linuxspi programmer from the current nightly build of AVRDUDE and found that https://github.com/kcuzner/avrdude/issues/10 does not occur. So my diverging commit https://github.com/facchinm/avrdude/commit/aa0d28f774147fb03f8b18cd065586c7a93a2b0b can safely be disregarded for the purposes of the upstream sync initiative.

## Migration

I found there is a breaking change to the format of the `-P` flag. The port format supported by the linuxspi programmer of Arduino's AVRDUDE is `/dev/spidev0.0`, while the upstream linuxspi uses the format `/dev/spidev0.0:/dev/gpiochip0`.

The formats are incompatible in either direction:

Using upstream AVRDUDE with the port format supported by Arduino's AVRDUDE:

```text
$ avrdude -p atmega328p -c linuxspi -U "flash:w:blink.hex:i" -P /dev/spidev0.0
avrdude: error: Unknown port specification. Please use the format /dev/spidev:/dev/gpiochip[:resetno]
avrdude: opening programmer "linuxspi" on port "/dev/spidev0.0" failed

avrdude done.  Thank you.
```

Using Arduino's AVRDUDE with the port format supported by upstream AVRDUDE:

```text
$ avrdude -p atmega328p -c linuxspi -U "flash:w:blink.hex:i" -P /dev/spidev0.0:/dev/gpiochip0

avrdude: error: Unable to open SPI port /dev/spidev0.0:/dev/gpiochip0
avrdude: error: Unable to open SPI port /dev/spidev0.0:/dev/gpiochip0

[...]

avrdude: error: Unable to open SPI port /dev/spidev0.0:/dev/gpiochip0avrdude: error: AVR device not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
```

MCUdude, 2022-04-11:
Thanks for the information @per1234! Avrdude is written in "classic" C, so the code is fairly easy to read and understand IMO.

> I found there is a breaking change to the format of the -P flag. The port format supported by the linuxspi programmer of Arduino's AVRDUDE is /dev/spidev0.0, while the upstream linuxspi uses the format /dev/spidev0.0:/dev/gpiochip0.

I don't think it would be easy to convince Jörg to accept `/dev/spidev0.0` as a valid SPI port, but I don't think it's really a problem since `/dev/spidev0.0:/dev/gpiochip0` can be used. AFAIK Arduino doesn't prove an official solution that relies on `linuxspi` for uploads. If we are able to get a default SPI port into Avrdude, the `-P` wouldn't even be necessary if using a Raspberry Pi.

mirh, 2023-05-18:
I *think* you can read the latest (and briefest) on the matter over at https://github.com/arduino/avrdude-build-script/issues/15#issuecomment-1541556556
Maybe you could even get some of this forever needed testing going.

MCUdude, 2023-05-19:
I'm closing this issue as the Arduino Avrdude "fork" has been moved over to [arduino/avrdude-build-script](https://github.com/arduino/avrdude-build-script)

closed 2023-05-19 by MCUdude